Short-bus/pilomar #56: Initial menu does not recognize the 'x' key

tproffen, 2024-02-04:
It looks like when starting pilomar the very first menu does not recognize the 'x' key and starts 'try again'.   Here is the terminal dump

`pi@pilomar:~/pilomar/src $ ./pilomar

SoftwareStartDatetime: 2024-02-04 20:09:45.032943+00:00 UTC
Using realtime clock.
Project root /home/pi/pilomar
Current time is: 2024-02-04 20:09:45.033469+00:00  UTC, offset is None seconds.
pilomar.py 2024-01-30 19:18:18.970106
Main log to /home/pi/pilomar/log/pilomar_20240204200945.log
Camera log to /home/pi/pilomar/log/pilomar_camera_20240204200945.log
Removing out of date log files to preserve disc space...
find /home/pi/pilomar/log -type f -name 'pilomar_*.log' -mtime +2 -delete
Done.
2024-02-04 20:09:45 Loading parameters from file: /home/pi/pilomar/data/pilomar_params.json
Current UTC time is: 2024-02-04 20:09:45.283973+00:00
Local time is: 2024-02-04 15:09:45.283973-05:00
Python3 is UTF-8 compliant, make sure that the terminal is in UTF-8 mode too.
2024-02-04 20:09:45 Initial: Using SD card for images. /
Motors enabled
DetectCamera: raspistill -o /home/pi/pilomar/temp/testraspistill.jpg
WARNING: reported in LogFile: 2024-02-04 20:09:45 DetectCamera: Camera not found.
DetectCamera: False
Camera disabled
2024-02-04 20:09:45 The program will generate simulated images instead.
2024-02-04 20:09:48 azimuth motor restored to last known position 48000 (  180.000° )
2024-02-04 20:09:48 altitude motor restored to last known position 0 (  000.000° )
2024-02-04 20:09:48 Home location Latitude 36.017935 N, Longitude 84.215594 W
2024-02-04 20:09:48 Hipparcos data cache exists, using that.

 Select target 
 1 Resume last observation      2 Repeat earlier observations 
 3 Solar system object          4 Hipparcos star catalog      
 5 Messier catalog              6 New General Catalog (NGC)   
 7 Comet                        8 Meteor shower               
 9 Space stations/satellites   10 RA-DEC co-ordinates         
11 Fixed ALT-AZ point          
 x Exit                         ? Refresh                    
Menu option : x
? Try again.

 Select target 
 1 Resume last observation      2 Repeat earlier observations 
 3 Solar system object          4 Hipparcos star catalog      
 5 Messier catalog              6 New General Catalog (NGC)   
 7 Comet                        8 Meteor shower               
 9 Space stations/satellites   10 RA-DEC co-ordinates         
11 Fixed ALT-AZ point          
 x Exit                         ? Refresh                    
Menu option : `

Short-bus, 2024-02-05:
Well spotted, I'll check that out. 
As a workaround you can just select '_1 Resume last observation_' to get out of the menu and keep the current selection.
When the program first starts up you have to select a target in order to proceed, but after that it should allow you to back out of target selection and keep the current choice.

Short-bus, 2024-02-05:
Testing solution which reloads the 'last target' automatically if you 'x' out of the target selection menu.
Will be part of the 2024-01-issues branch when ready.

Short-bus, 2024-02-27:
Done.

closed 2024-02-27 by Short-bus