Short-bus/pilomar #59: ProjectRoot not fully respected

Marcus8675, 2024-02-08:
Noticed at least 1 area in pilomar.py where the new ProjectRoot is not utilized.

FolderList["dataroot"] = "/home/pi/pilomar/data/" # Create default DATA pointer.
FolderList["log"] = "/home/pi/pilomar/log/" # Create default LOG pointer.

Short-bus, 2024-02-09:
Thanks Marcus, This should be fixed in the 2024-01-issues branch.
If it's critical for you, just alter your copy for now.

```
FolderList={} # Dummy entry until FolderList is fully defined. astrocamera doesn't initialize otherwise. 
FolderList["dataroot"] = ProjectRoot + "/data/" # Create default DATA pointer.
FolderList["log"] = ProjectRoot + "/log/" # Create default LOG pointer.
```

You can copy the latest code from [2024-01-issues](https://github.com/Short-bus/pilomar/tree/2024-01-issues) too of course - but you'll need to pull ALL the new codes from /src/*.py and also  update /circuitpython/code.py onto the microcontroller.

Thanks for the heads up about it.

Marcus8675, 2024-02-18:
I think I found another hidden spot.  line 1240 in pilomar.py

USBDiscMonitor = discmonitor(name='usb',devname=usbdev,path='/media/pi',disctype='usb',logger=MainLog.Log) # Create USB memory card monitor (if it exists).

If not installed in "/pi" folder then the USB mount will fail with '/media/pi' path

Short-bus, 2024-02-21:
Added 'path' parameters to the parameter file to allow for different mount paths.
Parameters.SDPath = '/' for monitoring the root SD card storage.
Parameters.USBPath = '/media/pi' for monitoring USB connected storage.
If you use different mount points now you can change the above parameters in the parameter file to retain your chosen paths across future software versions.

Have also made a note that the logic for finding the USB storage can be simplified in future. If the RPi starts up the desktop when it boots then the mounting is done automatically and more reliably. The current logic tries to find and mount USB storage even if there is no desktop running, but it is over-complex compared to having the desktop launched.

Short-bus, 2024-02-27:
Keeping this one open after the latest updates because I want to review the logic and simplify it further. But there is now a workaround in place for USB storage which is mounted under alternative paths.

Marcus8675, 2024-02-28:
May want to check your build scripts also for the new path parameters.  I think I saw some fixed paths in there.

Short-bus, 2024-07-10:
2024-04 Development branch now merged.
Closes this issue.

closed 2024-07-10 by Short-bus