Short-bus/pilomar #61: Scheduled Observations

Marcus8675, 2024-02-23:
Enhancement feature.  Options to schedule observations.  Foe example, ISS will be over my area from 8:42-8:53pm tonight.  I want to set up the system, tune and home all the motors, and schedule it to automagicly start the observation at the correct time.  Bonus if I can plan a full night of observations on multiple targets at various times.  would be a nice scheduling feature.  I could just follow up at the end of the evening with bias, dark and dark flats.

Short-bus, 2024-02-23:
I imagine it can be done. It would require a few new features I guess.

- A way to maintain an observation queue. (The solution used for recording 'previous targets' may be the basis?)
- A parent loop for the main ObservationRun routine to loop through the queue and switch target as needed. (Probably OK, I tried a 'survey' function which does something similar in an experimental program).
- A cut-off clock for ObservationRun to quit each observation at the right time. (Could be an attribute of the target when it's selected)

When it comes to imaging the ISS are you thinking to keep the ISS centre of the image, or park the telescope at just the right spot to catch it passing through the field of view?

Parking requires a simple ALT-AZ target to be used - you 'just' have to calculate the optimum ALT-AZ locations. I think the skyfield package can help with that calculation.

Tracking is more tricky: The ISS motion is at the upper limit of the telescope's speed so that may have an impact somewhere. Also with short lived time critical targets like that you need to consider the 'cycle time' between successive images. _This is similar to another builder's question about imaging eclipses._ It may be necessary to strip out all unnecessary tasks from the observation for the duration of the pass so you get the maximum number of images caught. I suspect you'd need some sort of 'loiter' function so that the telescope can pre-position itself ready for the ISS to pass into the field of view. So something special in the way the trajectory is calculated?

Marcus8675, 2024-02-23:
The ISS was just an example (and a bad one) but I guess you could park it and image it like meteor detection.  

I do like how you are already envisioning the implementation structure.  Let me try to explain the ideal end state that I had in mind.  Note that I absolutely do not expect this to be achieved quickly or at all but you know...to reach the clouds you have to aim for the stars.  :)

I was envisioning a final state where the PiLomar is outside for extended periods, if not in a permanent installation.  Remotely, I can schedule observations.  Love to have a calendar of upcoming solar events (comet xyz on Tuesday, Planet pdq rises on Thursday, etc..) and be able to plan my daily/weekly observations from the warm comfort of 'remote location'.  Starting to think about how to automate lens caps for end of session and dark files.  But this step is really about the first baby step to the end state.  Lets say on Friday I want to image Jupiter for the first 2 hrs, and then once that is past its prime viewing area, I would transition to Orion Nebula for an hr or so.  I can see a world where I spend my Saturday stacking and processing images from the prior week discounting the cloud covered days.

Short-bus, 2024-07-07:
This isn't implemented yet, but the next release has some preparation changes.
I've made a more general purpose solution to the 'target history' handling.
The format is more flexible so can also be used to handle lists of observations for other purposes.

2 additional uses come to mind at the moment.
1) The 'schedule' that you have raised here.
* I still need to think about how to achieve a simple and reliable solution to this one. But I really like the idea.
2) A list of 'suggested targets' that could be calculated.
* The software could run through the planet, comet, satellite, messier and NGC catalogs and list the brightest or largest targets that will be visible for your location. I wouldn't make it run through the Hipparcos catalog though... 100,000 stars could take a while :)
* I still need to consider an efficient way to filter these lists.

In the next release it's only used for remembering recent observations.

closed 2025-04-20 by Short-bus