Short-bus/pilomar #66: On chip cleanup not functional in new Bookworm build

Marcus8675, 2024-03-10:
I was unable to disable the on chip cleanup using the new build on a Bookworm system.  Changing the setting to false in the pilomar.json file had no effect.  Upon boot on chip cleanup was listed as ENABLED.  Using the camera menu to disable this feature also seems to have no effect.  When camera information is displayed from this menu if always shows ENABLED.  The main capture window also shows ENABLED.

Short-bus, 2024-03-10:
Hi. Well spotted, I'll need to clean up the UI a bit for that. Good to have a second pair of eyes running over it all.

Under Bookworm the camera is run by libcamera and that now has the 'denoise' option when you call it.
So the libcamera command templates (in the parameter file) have the '--denoise off ' option set in each case.
As a result the earlier onchip cleanup solution is only used with Buster builds.
So the onchip cleanup references are misleading under Bookworm because it's handled differently. I'll need to do some tidying up, I'd not noticed that inconsistency earlier, but it makes sense to hide that misleading info on a Bookworm install.

With this latest version I've moved more camera options into the command templates instead of having them hardcoded into the program. So hopefully overall it's easier to play with different camera settings as needed.

Short-bus, 2024-03-11:
Testing modified logic. 

### In Buster: 
The original DisableCleanup parameter and UI references work as before.

### In Bookworm: 
The DisableCleanup parameter will be removed. 
The UI references now check the CameraLightCommand parameter to see if the '--denoise off' parameter is set. 
If you try to enable/disable cleanup via the menu options it will recommend that you edit the command templates in the parameter file.
To enable on-chip cleanup, remove the '--denoise off' item from the command templates.
To disable on-chip cleanup, add the '--denoise off' item to the command templates.
The default templates have '--denoise off' in place. So you have to choose to turn on-chip cleanup back on.

Short-bus, 2024-07-10:
2024-04 Development branch now merged.
Closes this issue.

closed 2024-07-10 by Short-bus