Short-bus/pilomar #74: NGC catalog build fails if user is not 'pi'

Short-bus, 2024-05-29:
Raised by unhuzpt

Is this something you already know about?
I am running in bookworm with full install :)

`Camera enabled
2024-05-28 10:25:28 No recovery log file found for azimuth motor. Assuming  180.000Â°
2024-05-28 10:25:28 No recovery log file found for altitude motor. Assuming  000.000Â°
2024-05-28 10:25:28 Home location Latitude 49.781880 N, Longitude 6.225176 E
2024-05-28 10:25:28 Hipparcos data cache exists, using that.
[#################################] 100% de421.bsp
2024-05-28 10:25:35 Loading Messier catalog from /home/unhuzpt/pilomar/data/messierobjects.json ...
[#################################] 100% constellationship.fab
2024-05-28 10:25:36 Loading constellation patterns...
2024-05-28 10:25:36 Loading New General Catalog (NGC) entries from /home/unhuzpt/pilomar/data/ngc.json ...
2024-05-28 10:25:36 No NGC data cache, creating one now.
Exception in thread Thread-1 (StartMctlComms):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/unhuzpt/pilomar/src/pilomar.py", line 3472, in StartMctlComms
    Mctl.CommsLoop(UartControlQueue)
  File "/home/unhuzpt/pilomar/src/pilomar.py", line 3404, in CommsLoop
    self.ReadPoll() # Read anything waiting in the input buffer.
    ^^^^^^^^^^^^^^^
  File "/home/unhuzpt/pilomar/src/pilomar.py", line 3226, in ReadPoll
    while self.uart.in_waiting: # Something in the read queue.
          ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 549, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 5] Input/output error
2024-05-28 10:26:12 Loading meteor shower list from /home/unhuzpt/pilomar/data/meteors.json ...
2024-05-28 10:26:12 Loading comet list from https://www.minorplanetcenter.net/iau/MPCORB/CometEls.txt ...
[#################################] 100% CometEls.txt
2024-05-28 10:26:14 Loading CelesTrak station data from https://celestrak.org/NORAD/elements/gp.php?GROUP=stations&FORMAT=tle
You must select a target to start the program.

 Select target 
 1 Resume last observation      2 Repeat earlier observations 
 3 Solar system object          4 Hipparcos star catalog      
 5 Messier catalog              6 New General Catalog (NGC)   
 7 Comet                        8 Meteor shower               
 9 Space stations/satellites   10 RA-DEC co-ordinates         
11 Fixed ALT-AZ point          
 x Exit                         ? Refresh                    
Menu option : 

`

_Originally posted by @unhuzpt in https://github.com/Short-bus/pilomar/discussions/72#discussioncomment-9579461_

Short-bus, 2024-05-29:
Additional info from unhuzpt:

So......
Yeah, the "issue" was that your script assumes the user is 'pi' and that was not my case :)
So tired lately that i forgot this when i build the 1st time and noticed it....
Now it installed all fine and is working as expected :)

Allow me to suggest maybe to adapt the script to something like:

#Get current user
username=$(whoami)

#use current user
export pilomarhome="/home/${username}/pilomar"

Or maybe even ask if one wants to use the current user or another?

Short-bus, 2024-05-29:
The program should adapt to different users, so there's a bug somewhere.
Needs investigating.

unhuzpt, 2024-05-30:
I think for my case there were 2 points that made it work:

- i forgot to disable the console to serial port
- i had to manually edit the file scripts/buildpilomar_bookworm64 

In the installer script the first lines were with user pi so would not create the correct folders and place files on the right place:

export pilomarhome='/home/pi/pilomar'
echo home directory is: $pilomarhome
export pilomarlog='/home/pi/pilomar/log/buildpilomar_bookworm64.log'
echo build logfile is: $pilomarlog

Now as i am under a lot of fatigue this week i really do not remember the sequence of events.
I think after the errors on the installation, i ended up on changing the  file "file scripts/buildpilomar_bookworm64" replacing pi by unhuzpt and then it installed correctly. And maybe only afterwards i disabled the console to serial output.
But i may be wrong.

[pilomar_20240528102522.log](https://github.com/Short-bus/pilomar/files/15493836/pilomar_20240528102522.log)
[pilomar_20240528093337.log](https://github.com/Short-bus/pilomar/files/15493838/pilomar_20240528093337.log)
[pilomar_20240528093140.log](https://github.com/Short-bus/pilomar/files/15493845/pilomar_20240528093140.log)
[pilomar_20240529161918.log](https://github.com/Short-bus/pilomar/files/15493864/pilomar_20240529161918.log)
[pilomar_20240529161918.log](https://github.com/Short-bus/pilomar/files/15493869/pilomar_20240529161918.log)

unhuzpt, 2024-05-30:
I found a cell picture i took to look into the issue, this was on the 27th and the log dir only has 28 and 29 so here it is where is clear things went right after i edited the installer script.
[image]

Short-bus, 2024-05-30:
Great, thanks, I'll fix those for the next release.
Plus: If I can trap the UART error it can make a suggestion to check the serial console too.

unhuzpt, 2024-05-30:
Something like:

# Create communication with microcontroller over uart0 serial port.
    mctl = microcontroller(port='/dev/serial0',resetpin=Parameters.MctlResetPin,boardtype=Parameters.BoardType)
except PermissionError as e:
    print(f"Error: {e}. Please check the permissions of /dev/serial0.")
    log.error("Failed to open serial port due to permission error")

Maybe can improve this if i take some time to look at the code with more time :)

Short-bus, 2024-07-07:
I've added the specific PermissionError exception check to the code, with an explanation about the SERIAL port configuration.

Short-bus, 2024-07-10:
2024-04 Development branch now merged.
Closes this issue.

closed 2024-07-10 by Short-bus