Short-bus/pilomar #75: Closed door possibility?

kf6spf, 2024-07-06:
Hi ...

A fascinating project that does just about what I was looking for. Great work.

My big question is how the (fixed) dome handles weather? I'm here in "sunny" California, but even we get weather sometimes. How would this dome protect the camera and electronics from wind and water damage?

I have an AllSky camera set up running now, and have had to disable daytime images because the Pi4b running it gets way too hot. I'll have to redesign my pier mount to add fans and heat dissipation somehow. So the second question is how well does this handle excess heat and how would/should one dissipate it?

Michael

Short-bus, 2024-07-07:
Thanks Michael,

The current design definitely isn’t weatherproof enough. I think at least one other builder is looking at an evolution of the project that would be more sealed.

The most obvious gap being the lack of doors, but there are also other gaps to close off. It could all be done. I’ve not tried that yet. I don’t know if the bearing ring would need protecting too? I’m near the sea, everything corrodes here eventually.

Once the dome is sealed you are right there’s a heat problem. The original design uses passive cooling because we have cold enough nights up here. I’m currently experimenting with a RPi5 and I’m using the combined heatsink and fan on that.
For me the captured heat in the dome is a good microclimate to reduce condensation, but it’s a double edged sword.

For heat protection during the day, maybe the dome could be silvered or insulated?

Another thought, forget sealing the dome entirely and build a larger outer cover more simply? Uglier, but practical.

I’ve seen a revised camera cradle design that seals the camera board and adds a heatsink. The RPi and motorcontroller board could be relocated to a cooled and sealed container, that may be simpler than adding cooling and vents to the dome.

It’s possible to add doors in a few ways I guess. Either to make working sliding doors just like a full scale dome, or maybe have a fixed cover that the dome could rotate up against when not in use? (That would block the view of the sky where the cover is fixed, but would not require any additional electronics or motors.) Making sliding working panel doors would look fantastic, but could be an engineering challenge, probably needing a rethink of the microcontroller and PCB.

The published project doesn’t have weather detection, but it could be added. Humidity, wind and rain detection should be possible, there are plenty of free GPIO ports free on the RPi. I made a utility routine that uses metcheck.com’s json feed to give a ‘forecast’ and plan my observations around that. They provide weather and astro forecast web services for free. I’ve found them remarkably reliable.

Also, choose the materials wisely of course. I use PLA+ which is stable enough for our climate, but in high heat I think you need to use a more temperature stable material.

There are many directions to take the project, currently I’m concentrating on the image handling side. It’s going slowly because we have had unbelievably cloudy nights for about a year, and the summer nights are short and light.

If you re-post the question on the [Instructables project](https://www.instructables.com/Pi-lomar-3D-Printed-Working-Miniature-Observatory-/) page, there may be ideas from others too. 

Matt

DragonPrincess78, 2024-07-08:
Hi Michael,
what you are asking is exactly the project we are working on. Electric doors and rain detection.
The door mechanism itself is however not easy to build. Considering the size, you will need guide rails on top and bottom. We've got linear steppers on the bottom, and even with 5mm PETG with 35% infill the shutter doors have too much flex and get stuck. I'm currently abroad for work, but once I'm home again, I'll send you photos.
Rain sensing kind of works... we've got those rain drop sensors, which do sense drops, but it takes a lot to get them to engage properly. The camera itself is rainproved with laser cut eva foam and a housing. The control pi got its own housing in the back, with air intakes rain protected too. The only thing which is still in the open are the steppers.
This should be okay, but you will need to clean the lense after each rain anyway.

Material wise, I suggest using PETG or infused filament, maybe HIPS. PLA+ or ABS do not even work in UK summer from experience.